tapparelj/gr-lora_sdr #104: Circular import problems

radio-satellites, 2024-08-23:
Hello!

I feel like I have solved this issue before, but I'm not sure how to do it now. How do I resolve the circular import issue in `lora_sdr/lora_sdr_lora_tx.py` and `lora_sdr/lora_sdr_lora_rx.py`? The specific offending line is `from . import lora_sdr_python as lora_sdr`. 

Full error message is below. 

Thanks!

```Traceback (most recent call last):
  File "/home/ve3svf/OS42GroundSoftware/DEMOD.py", line 27, in <module>
    import gnuradio.lora_sdr as lora_sdr
  File "/usr/local/lib/python3.10/site-packages/gnuradio/lora_sdr/__init__.py", line 24, in <module>
    from .lora_sdr_lora_tx import lora_sdr_lora_tx
  File "/usr/local/lib/python3.10/site-packages/gnuradio/lora_sdr/lora_sdr_lora_tx.py", line 16, in <module>
    from . import lora_sdr_python as lora_sdr
ImportError: cannot import name 'lora_sdr_python' from partially initialized module 'gnuradio.lora_sdr' (most likely due to a circular import) (/usr/local/lib/python3.10/site-packages/gnuradio/lora_sdr/__init__.py)
```

radio-satellites, 2024-08-31:
@tapparelj sorry for the late reply! I didn't see it in time. 

Nope, this is in general when I try to run any flowgraph with lora_sdr in it. I'm running on Fedora 40.

LuccoJ, 2024-09-24:
Same here, also with Fedora 40, with gr-lora_sdr installed in /usr/loca.

radio-satellites, 2024-09-25:
This is almost definitely a Fedora issue, I've never had that in Ubuntu or Debian.