tapparelj/gr-lora_sdr #105: tx_rx_functionality_check.py fails with installation instructions.

sytang-unc, 2024-08-27:
Get the following when executing tx_rx_functionality_check.py as suggested at the end of the installation instructions.

```
modulate :info: set_min_output_buffer on block 3 to 10000000
Traceback (most recent call last):
  File "/home/sytang/gr-lora_sdr/examples/tx_rx_functionality_check.py", line 232, in <module>
    main()
  File "/home/sytang/gr-lora_sdr/examples/tx_rx_functionality_check.py", line 210, in main
    tb = top_block_cls()
  File "/home/sytang/gr-lora_sdr/examples/tx_rx_functionality_check.py", line 79, in __init__
    self.blocks_throttle2_0 = blocks.throttle( gr.sizeof_gr_complex*1, samp_rate, True, 0 if "auto" == "auto" else max( int(float(0.1) * samp_rate) if "auto" == "time" else int(0.1), 1) )
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.blocks.blocks_python.throttle(itemsize: int, samples_per_sec: float, ignore_tags: bool = True)

Invoked with: 8, 500000, True, 0
```

Looks like the version of gnuradio requested in environment.yml (3.10.3.0) doesn't accept the fourth argument to throttle. Looks like this was changed after 3.10.6.0.

tapparelj, 2024-08-27:
Hello @sytang-unc,
good catch! 
I will update the environment file to gnuradio 3.10.6. 
In the mean time, you can either remove the parameter from the python file or open the corresponding .grc script in gnuradio-companion and build/execute it from there.

sytang-unc, 2024-08-27:
I'm not understanding the `if "auto" == "auto"` and `if "auto" == "time"` in the argument. Why are these not always true and false?

tapparelj, 2024-08-28:
The python script is generated from [this](https://github.com/gnuradio/gnuradio/blob/main/gr-blocks/grc/blocks_throttle2.block.yml) gnuradio-companion block (line 61). I'm not sure the reason why the person that wrote that code didn't directly write the correct option in the python file.
In your case you can replace the throttle instantiation by this function : blocks.throttle(${type.size}*${vlen}, ${samples_per_second},${ignoretag})

tapparelj, 2024-10-04:
As it has been some time without follow-up I guess that this answered your question. Don't hesitate to reopen this issue if needed.

closed 2024-10-04 by tapparelj