tapparelj/gr-lora_sdr #108: Issue with number of inputs to frame_sync block

ask-gh, 2024-09-27:
Hi,

Firstly, thanks for all the hard work!

I have managed to get everything up and running and working smoothly apart from one issue. Due to the number of inputs defined in the forecast of the frame sync block:

https://github.com/tapparelj/gr-lora_sdr/blob/b53a3ba434df3150f4fb51a3e4bb1226caa51717/lib/frame_sync_impl.cc#L121

when using larger SF values and/or lower BW (and hence a larger os_factor) the number of inputs becomes too large and I get the error message:

[image]

For reference, I am using the RTLSDR and the soapy RTLSDR source block.

Everything seems to working absolutely fine when the value of `ninput_items_required[0] < 8192` and also the rx_tx_functionality_check.grc works perfectly.

Any help with this would be much appreciated. Thanks

tapparelj, 2024-10-04:
Hello @ask-gh,
This issue appears due to the default buffer size that GNU Radio allocates between blocks.
You should be able to fix it by checking #87 where I give more details.

Don't hesitate to let me know if you still can't fix it.

closed 2024-10-29 by tapparelj

ask-gh, 2024-11-21:
Thanks, this worked a charm!