tapparelj/gr-lora_sdr #116: Issue on installation on Mac3

Maryam919, 2024-10-27:
Hi,
I followed step by step the installation part and install gr-lora-sdr module (without condo), when I check I see lora_sdr_puthon.so, and saw the lora blocks inside GNURadio but when I open an example from your repository  it does not work! I trouble shout several times but it still does not work, last time all lora blocks disappeared from GNURadio!
Could you help me install is completely on Mac?

tapparelj, 2024-10-29:
Hello @maryam,
I've never tried the installation from source on Mac. If the block don't appear but you managed to build and install the oot module without errors, I would suspect an issue with the installation path. How did you install gnuradio? When installing this OOT module, what was installation path? If you open a terminal, run python and try to ''import lora_sdr'' does it find the module?

Maryam919, 2024-11-04:
Hi,
Thank you for your email, I sent the explanation on my email, this is
whatever i receive :

xxxx.....MacBook-Pro gr-lora_sdr % python3 -c "from gnuradio import
lora_sdr"

Traceback (most recent call last):

  File "<string>", line 1, in <module>

ImportError: cannot import name 'lora_sdr' from 'gnuradio'
(/opt/homebrew/lib/python3.12/site-packages/gnuradio/__init__.py)

xxxxx-MacBook-Pro gr-lora_sdr %
Sincerely,
Maryam

Maryam919, 2024-11-04:
***@***.*** build % ls
/opt/homebrew/lib/python3.12/site-packages/gnuradio/

__init__.py eng_option.py plot_psd_base.py

__pycache__ fec pyqt_filter.py

analog fft pyqt_plot.py

audio filter qtgui

bindtool gr soapy

blocks gr_unittest.py trellis

blocktool grc uhd

channels modtool vocoder

digital network wavelet

dtv pdu zeromq

eng_arg.py plot_data.py

eng_notation.py plot_fft_base.py
***@***.*** build %

 Sincerely,
Maryam

tapparelj, 2024-11-05:
I would repeat my question: when installing this OOT module, what was installation path? (It will be written in the terminal when you run ```make install``` while in the build folder.

Maryam919, 2024-11-06:
Sorry for confusion,
When I run make install on Mac:

-- Build files have been written to: /Users/maryamf/gr-lora_sdr/build

***@***.*** build % make install

[ 46%] Built target gnuradio-lora_sdr

[ 48%] Built target doxygen_target

[ 48%] Built target copy_module_for_tests

[ 51%] Built target extracted_docstrings

[ 53%] Built target lora_sdr_docstrings

[100%] Built target lora_sdr_python

Install the project...

-- Install configuration: "Release"

-- Up-to-date: /usr/local/include/gnuradio/lora_sdr/api.h

CMake Error at include/gnuradio/lora_sdr/cmake_install.cmake:41 (file):

  file INSTALL cannot set permissions on

  "/usr/local/include/gnuradio/lora_sdr/api.h": Operation not permitted.

Call Stack (most recent call first):

  cmake_install.cmake:42 (include)

make: *** [install] Error 1

***@***.*** build %

*Another question: *I was disappointed to install on Mac so I installed the
LoRa blocks from gr-Lora-sdr Github on the Ubuntu arm based VM.
I am really thankful if you could  clarify for me:
How can I install a Heltec LoRa32 (863~928) as a LoRa transmitter node to
send data and check the log file on the receiver side (my receiver is an
USRP) by GNU radio?
Sincerely,
Maryam

tapparelj, 2024-11-06:
Ok for the installation, there is an error for the installation path (I'm really not familiar with mac so I'll try my best).
Based on the error messages I would try to install the module where python looks for it, can you try to install the OOT module in /opt/homebrew/ by running 
```cmake .. -DCMAKE_INSTALL_PREFIX=/opt/homebrew```
```make install``` (you may need to use "sudo make install" to have the right to install things there).

For your other question:
I don't have that exact board available, but I could use sx1262 without issues. You need to get a library that allow you to configure the LoRa chip and send messages, then set the same configuration (spreading factor, center frequency, bandwidth,sync word,..) for the USRP (in gnuradio) and for your node. I can't test it, but [this](https://github.com/HelTecAutomation/Heltec_ESP32/tree/master) and look at the loraBasic examples should do the trick.

tapparelj, 2024-11-06:
There should be a format error that only appears on mac.. can you try copy and past the content of the file in https://www.yamllint.com/ and see if it finds any error?

Unfortunately, I can't give you a full tutorial. The link is the reference code from Heltec. You should be able to find some tutorial for your specific board.

Maryam919, 2024-11-06:
Many thanks.
I did  copy and paste the content of the file in https://www.yamllint.com/,
and received : Valid YAML!
Sincerely,
Maryam

tapparelj, 2024-11-06:
Then, sadly, I only see two options:
- Generate a random flowgraph and try to see what seems different between the .grc file generated and the one in the example.
- Reproduce the Tx flowgraph from scratch, following the template shown below (you can use the hierachical block to make it easier, at least to try it out).
[image]

Maryam919, 2024-11-06:
Thank you, I will check it out.

Sincerely,
Maryam

Maryam919, 2024-11-06:
Hi,
I downloaded lora-RX flowgraph. It works on Ubuntu, one question I have is,
how can I see the chirps? I must add which block and where inside the
flowgraph?
Sincerely,
Maryam

tapparelj, 2024-11-07:
Hello,
you can see them by using a "QT GUI Waterfall". However this block has a relatively low resolution so you will be able to see the chirps clearly only for large SFs (e.g. sf12). You might want to decrease the update period as well.
Else you can just save the IQ samples to a file and load them with any programing language you like and plot the instantaneous frequency with some form of differential(unwrap(angle(IQ_samples))).

tapparelj, 2025-02-14:
Given that the question is not anymore related to the issue, I'll close this issue.

closed 2025-02-14 by tapparelj