tapparelj/gr-lora_sdr #117: Demodulator LORA 2.4GHZ SX128X

MVPGLOCK, 2024-10-27:
Добрый день , я собрал схему и записал сигнал с пульта управления дроном на HackRF частотой 2.4 GHz , увидел по спектру что передается LORA и хотел демодулировать , открыл рекомендации и хар-ки сигнала и вбил в схему 
[image]
Вот
[image]
Но при демодулировании выходит ошибка Header Checksum invalid 
[image]
Можете ли вы помочь . Просто я пытался ставить различные другие параметры и по другим рекомендациям какие только есть в интернете , но увы они ничего не отображали . Я пытался вводить допустим вот это SX128x - LoRa
Begin method:
Carrier frequency: 2400.0 MHz
Bandwidth: 812.5 kHz (dual-sideband)
Spreading factor: 9
Coding rate: 4/7, long interleaving disabled
Output power: 10 dBm
Preamble length: 12 symbols
Other:
LoRa header mode: explicit
LoRa CRC: enabled, 2 bytes
Inaccessible:
LDO regulator mode: disabled (SX128x module with DC-DC power supply)
Number of CAD symbols: 8 
Но безуспешно и сейчас уже более менее что то появилось после того как я по описанной выше рекомендации ввел , но все же столкнулся с ошибкой . Мой пульт это Literadio 2SE -ERLS ,помогите пожалуйста

eabase, 2024-11-07:
Why is this in Russian? I guess if you can't be bothered to translate and post in English, anyone who can help you will probably also not bother...

H-a-r-sha, 2025-06-21:
Hi @MVPGLOCK,

Did you happen to resolve the issue you mentioned earlier? I'm facing a similar problem while capturing LoRa packets using my SDR. The chip in use is the SX1276, and I’ve configured the LoRa parameters correctly — including SF, BW, and other settings. However, I keep getting the same message repeatedly on the terminal: **"Header Checksum Invalid."**

I’ve tried adjusting parameters like switching between implicit and explicit header modes, toggling CRC and soft decoding options, but when I do, the `rx_msg` output appears as gibberish characters like **�!�!**, which are clearly invalid.

Could you please help me troubleshoot this or share how you resolved it if you managed to?

Thank you!