tapparelj/gr-lora_sdr #118: Are there any build instructions for using MSYS or MinGW64 on Windows?

eabase, 2024-11-02:
It seem to me that it should be possible to compile this using `MSYS/UCRT` (or MinGw64) for Windows, and the only hurdle may be crazy usage of `Conda`. As we all know `conda` is the mega bloated python package that includes just about every python package you can imagine. Clearly only a small percentage of these are actually needed for the LoRa functionality, but since developers are mostly lazy and just make conda a requirement, we end up with not only losing overview of minimal but necessary requirements, but also hugely inflating the working application, while limiting builds to *nix based OS's.

Has anyone attempted to build this natively for Windows, using MSYS (or VS)?

eabase, 2024-11-05:
@tapparelj 
Hi Joachim, 
> ... specifically developed for Linux, where building from source is a straightforward option. 

Great, but the very first part of "building from source" tell you to install `Miniconda`. That's not really building from source... 

Do you know, or is it documented somewhere, exactly what packages from Conda are used?

tapparelj, 2024-11-06:
Sorry if you got confused.
It was mentioning that "all dependencies can be installed with conda **or** you can install them individually".
I hope the new phrasing makes it less confusing for others.
Thanks to the contribution of @tinyboxvk the list of conda packages has been reduced to the minimum required packages.

eabase, 2024-11-07:
Hi,
I just watched your video at `GNURadioCon`. Very nice presentation! And very interesting other events. 

I have managed to build it in MSYS/MINGW64, and will provide all build instructions once I can verify it actually works, and also for using different build tools. In fact I am not sure how to run it, and get it to do something in GNU Radio. (I am new to GR.)

What files do I need to open, run, move?
(It's not clear as *ldconfig* is not used in MSYS, and GR locations are completely different.)

Can you tell me how to run a test of some kind?

tapparelj, 2024-11-07:
Hello,
ldconfig is not necessary anymore anyway.
there is a demo flowgraph that just connects the transmitter output to the receiver in examples/tx_rx_functionality_check.grc. If you open this file, in gnunradio-companion you should the the different blocks connected. 
If everything is fine you should be able to press the "play" button and you should see in the terminal window the content of the decoded packet being displayed.

I can only talk for linux, but the key locations that gnuradio need to know is where are the libraries ([this](https://wiki.gnuradio.org/index.php/ModuleNotFoundError) page might help you) and where are the gnuradio-companion block descriptions (to deiplay them in gnuradio companion). For the latter (for linux) you can set the path to blocks in ~/.gnuradio/config.conf if you install the module in a custom location with the lines
```
[grc]
local_blocks_path = <your custom install prefix>/share/gnuradio/grc/blocks```

eabase, 2024-11-08:
Strange, I don't have a `~/.gnuradio/config.conf`.

Instead I have the following: 
```bash
# gnuradio-config-info.exe --print-all

C:\msys64\mingw64\
C:\msys64\mingw64\etc
C:\msys64\mingw64\etc\gnuradio\conf.d
C:\msys64\home\xxxx\.config\gnuradio
C:\msys64\home\xxxx\.local\state\gnuradio

Sat, 14 Sep 2024 05:38:16Z
testing-support;python-support;post-install;doxygen;man-pages;gnuradio-runtime;common-precompiled-headers;gr-ctrlport;gnuradio-companion;JSON/YAML config blocks;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* portaudio;* windows;gr-channels;gr-pdu;gr-iio;* libad9361;gr-qtgui;gr-trellis;gr-uhd;gr-uhd UHD 4.0 RFNoC;gr-utils;gr_modtool;gr_blocktool;gr-video-sdl;gr-vocoder;* codec2;* freedv;* gsm;gr-wavelet;gr-zeromq;gr-network;gr-soapy

3.10.11.0

gcc.exe (Rev1, Built by MSYS2 project) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software see the source for copying conditions.  There is NO
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

g++.exe (Rev1, Built by MSYS2 project) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software see the source for copying conditions.  There is NO
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

D:/a/msys64/mingw64/bin/gcc.exe:::-O3 -DNDEBUG -march=nocona -msahf -mtune=generic -O2 -pipe -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wp,-D__USE_MINGW_ANSI_STDIO=1 -Wsign-compare -Wall -Wno-uninitialized -Wignored-qualifiers -Wcast-qual
D:/a/msys64/mingw64/bin/g++.exe:::-O3 -DNDEBUG -march=nocona -msahf -mtune=generic -O2 -pipe -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wp,-D__USE_MINGW_ANSI_STDIO=1 -Wsign-compare -Wall -Wno-uninitialized -Wignored-qualifiers -Wcast-qual

2.13.5

#--------------------------------------------------------------------
# tree -L 2 .config/

.config/
└── gnuradio
    └── grc.conf

#--------------------------------------------------------------------
# tree -L 3 .local/

.local/
└── state
    └── gnuradio
        └── grc.log

#--------------------------------------------------------------------
# cat .config/gnuradio/grc.conf

# This contains only GUI settings for GRC and is not meant for users to edit.
#
# GRC settings not accessible through the GUI are in config.conf under
# section [grc].

[main]
file_open =
main_window_width = 1402
main_window_height = 807
console_window_position = 484
blocks_window_position = 1160
variable_editor_position = 449

[files_recent]
files_recent_0 = C:\xxxx\yyyy\gr-lora_sdr\build\opt\share\gr-lora_sdr\examples\lora_RX.grc

[files_open]

```

eabase, 2024-11-08:
Loads, but doesn't look healthy. 

[image]

---

Also I get the following in my terminal:
```bash
#  gnuradio-companion.exe &
# ...
(gnuradio-companion.exe:10288): GLib-GIO-CRITICAL **: 13:50:49.815: GFileInfo created without standard::type

(gnuradio-companion.exe:10288): GLib-GIO-CRITICAL **: 13:50:49.815: file ../glib-2.82.2/gio/gfileinfo.c:
line 1610 (g_file_info_get_file_type): should not be reached
# ...
```

Adding the python exports, removed the error above:
```bash
# They were never set/defined
export PYTHONPATH=/c/msys64/mingw64/lib/python3.11/site-packages:$PYTHONPATH
export LD_LIBRARY_PATH=/c/msys64/mingw64/lib:$LD_LIBRARY_PATH
```

Adding the `~/.gnuradio/config.conf`: 
```bash
# cd ~
# mkdir .gnuradio && cd ./.gnuradio
# Paste all of the below.
# tee -a config.conf &>/dev/null <<EOF
#----------------------------------------------------------
# File		: config.conf
# Location	: ~/.gnuradio/config.conf
#----------------------------------------------------------
[grc]
#local_blocks_path = <your custom install prefix>/share/gnuradio/grc/blocks
local_blocks_path = /c/xxxx/xxxx/gr-lora_sdr/build/opt/share/gnuradio/grc/blocks

EOF
```

This seem to start the block automatically, but still look broken.
I need to look at this some more.