tapparelj/gr-lora_sdr #119: Unable to demodulate to/from x410 and other USRPs

jdloft, 2024-11-04:
First off, thanks for this awesome project! I appreciate the work you and others went through to get this working as well as it does.

I've been testing sending/receiving between a handful of USRP models and I've managed to get it working between a couple of N210s and a B200mini, but can't seem to get any packets correctly demodulated to/from our x410 and any other USRP model. Sending and receiving between different channels on the x410 works as expected, but I'm seeing some sort of timing issue to and from other devices. I can see the chirps coming into the RX chain correctly, so I know at least we're locked on the right channel.
[image]

With SF=7 I get corrupted headers and with any other SF I don't even detect headers.
```
----------------------------------------------------------------------
Tag Debug:
Input Stream: 00
  Offset: 22016  Source: n/a     Key: frame_info   Value: ((sf . 7) (cfo_frac . 0.158288) (cfo_int . 0) (is_header . #t))
----------------------------------------------------------------------

--------Header--------
Payload length: 15
CRC presence:   1
Coding rate:    2
Header checksum valid!

----------------------------------------------------------------------
Tag Debug:
Input Stream: 00
  Offset: 23040  Source: n/a     Key: frame_info   Value: ((ldro . #f) (symb_numb . 38) (is_header . #f) (err . 0) (crc . 1) (pay_len . 15) (cr . 2))
----------------------------------------------------------------------
rx msg: 0x4d, 0xe5, 0x66, 0x3d, 0xe, 0xed, 0x22, 0xdf, 0x10, 0x20, 0x50, 0x9b, 0xec, 0xb0, 0x34
CRC invalid

----------------------------------------------------------------------
Tag Debug:
Input Stream: 00
  Offset: 26880  Source: n/a     Key: frame_info   Value: ((sf . 7) (cfo_frac . 0.16465) (cfo_int . 0) (is_header . #t))
----------------------------------------------------------------------

--------Header--------
Payload length: 31
CRC presence:   1
Coding rate:    2
Header checksum invalid!
```

I've taken recordings of a packet from an N210 and x410 and they look nearly identical.

Any ideas?

tapparelj, 2024-11-05:
Hello @jdloft,
as the x410 works only with "itself", I would look if there is a difference of sampling rates. When running the flowgraph directly in a terminal to easily look at the log (not within gnuradio-companion), is there any information or warning about the selection of a clock frequency different that the one you request in the UHD USRP source/sink? (Maybe you can paste everything that get printed during the start of the flowgraph)

jdloft, 2024-11-08:
Turns out the DUC is not fine-grained enough to hit 250kHz and GR was rounding the actual TX rate. Thanks for looking into this!

closed 2024-11-08 by jdloft