tapparelj/gr-lora_sdr #122: RX Error

smsajjadi, 2024-11-18:
Hi

I have been trying to test this valuable code with a ADALM-pluto platform. First I tried to replace the sink from USRP to IIO-based pluto sink, but it didn't work. So, I used both the source code and data from a GNU-radio run, to implement a bit-accurate implementation of the code in python and use the ADI module to send it through ADALM-pluto. With this trick, I began  receiving some packets in an sx127x module. But unfortunately, this is not working as expected. I am sending exactly the same signal repeatedly, but I receive different data everytime, and RX Error is issued in every single packet. received data length is correct. 
I don't have another SDR to check receipt with that. 
Where should I start to debug this situation?

Spec of link: sf=7, explicit header, CR=1, 8 preambles.

smsajjadi, 2024-11-19:
Going forward, I switched to implicit header. Then I saw that the first 5 nibbles which are encoded with hamming(8,4) are OK most of the time.

Then I switched over to CR=4 and now 12 correct nibbles. (My data is 12 bytes). So it seems like that hamming encoding and maybe interleaving have some sort of bug here

smsajjadi, 2024-12-03:
I have found out that my switching to plutoSDR was the main point of failure. I had to (not yet knowing why) add enough padding zeros to end of transmission, to achieve correct receipt! The main code has no bugs whatsoever it seems. 

Thank you and sorry for the inconvience. Hope this tip help somebody some time.

tapparelj, 2024-12-04:
Sorry for arriving too late, but I'm glad you figured out how to solve your issue!

closed 2024-12-04 by tapparelj