tapparelj/gr-lora_sdr #125: RX Script Crashes with SF = 5 and SF = 6

gorsheep, 2024-12-17:
Hello,

First of all, thank you for providing this great project! I've been using it to successfully implement LoRa communication. However, I've encountered issues when using spreading factors (SF) 5 and 6. Specifically, the RX script crashes, and packets are not decoded correctly under the following settings:

Transmitter (TX):
- samp_rate: 500 kHz
- bw: 250 kHz
- TX_gain: 100
- center_freq: 2.1 GHz
- sf: 6
- cr: 1

Receiver (RX):
- samp_rate: 500 kHz
- bw: 250 kHz
- RX_gain: 100
- center_freq: 2.1 GHz
- sf: 6
- cr: 1

I’ve attached the following screenshots for reference:
1. TX flow graph
2. RX flow graph
3. Console output showing that packets are not decoded correctly.

I have tried increasing the sampling rates (to 2x, 4x, and 8x the bandwidth) for both TX and RX, but the issue persists. The RX script still crashes when using SF = 5 or SF = 6. Other SF values (e.g., SF = 7) work as expected with the same bandwidth and sampling rate.

I noticed that a similar issue was reported about a year ago. The authors mentioned creating a new branch to address it, but I couldn’t find clear documentation or confirmation on whether this fix has been merged into the main branch, which I am currently using.

Could you please clarify if this issue has been resolved in the main branch? If not, are there any specific changes I should test from the branch created to address this problem?

Thank you for your time and assistance. Please let me know if further details or testing are needed!

[image]
[image]
[image]

jdloft, 2025-01-08:
This looks exactly like what I was seeing with #119. Make sure your radio is actually hitting the sample rates you're requesting. In my case one of the radios wasn't able to hit 250kHz.

Also, the script is crashing when it tries to decode a corrupted packet, as you can see with the header checksum invalid message. You can change the verify CRC block to not print the message in ASCII to fix this.

tapparelj, 2025-01-09:
Hello @gorsheep,
the current main branch only supports the spreading factors 5 and 6 in implicit header mode (i.e. no explicit PHY header). This was the was LoRa chips version sx1276 and earlier where working.
Newer version of LoRa chips have added support for explicit headers with sf 5 and 6. The branch _sf5_6_sx126x_  works fine for transmissions SDR -> SDR or Semtech chip -> SDR, but SDR -> Semtech chip currently fails. As one type of transmission is not working, it has not been merged to the main branch yet. 

For the crash, gnuradio-companion freezes when trying to print some special character in the integrated terminal. You can either follow the trick of @jdloft  or directly execute the python script (generated by gnuradio-companion) in a terminal.

tapparelj, 2025-01-14:
Hello,
with impl_head=False, there will be an "explicit" header prepend to your payload and contains information about the coding rate, the presence of a CRC, and the length of the payload. With the implicit header set to true, the transmitter will not include any of those information and assume  the receiver knows those parameters. 
The receiver needs to know the length of the frame for 2 purposes:
- To not waste time and energy demodulating noise
- To know where the payload CRC is located to be able to check the frame integrity

If you don't care about the CRC verification and power consumption is not too much of a concern, you could set the receiver to a large frame length (e.g., the maximum frame length: 255 bytes) and add yourself a termination character or sequence that you can use to know what is your data and what is just random noise in the decoded message.

You can have a look at [this paper](https://pubs.gnuradio.org/index.php/grcon/article/view/145/123) for more details on LoRa and this implementation. And don't hesitate to tell me if it is still confusing.

gorsheep, 2025-01-14:
Now I understand, thank you very much!

closed 2025-01-15 by tapparelj