tapparelj/gr-lora_sdr #129: problem about output

Remain123, 2025-02-13:
Hi, I’m a student currently using your library for some experiments. I’ve fully installed it on my Raspberry Pi, which is connected to my USRP B210. I used a Seeeduino LoRaWAN as the signal transmitter. I successfully received the “hello world” message that I sent. I then placed File Sink blocks after the USRP Source and after the Frame Sync, respectively, and saved the resulting data to “output.bin.” After converting that file to a text format and plotting it in MATLAB, I noticed that the waveform is clearly different from a correct LoRa signal. I need the proper signal for further work. Could you please point out what might be going wrong? I’ve attached my GNU Radio flowgraph, along with the plots of the data captured after the USRP Source and after the Frame Sync, respectively.
Thanks！

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

tapparelj, 2025-02-14:
Hello @Remain123, the signal you show on your last figure look pretty much like the real part of a LoRa chirp.
Where you expecting to see something like this?
[image]
If yes, it is the instantaneous chirp frequency that looks like that. And you can get it with something like inst_freq=unwrap(diff(angle(complex_samples)))

Remain123, 2025-03-06:
Dear tapparelj, thanks for your reply.
According to your suggestion, I have plotted the time-frequency spectrogram of the obtained data.

These two images are the time-frequency spectrogram and the LoRa signal plot obtained from the data output by the ”File Sink“ connected after the USRP Source.
The obtained LoRa signal differs significantly from the well-known LoRa signal. However, the time-frequency spectrogram shows noticeable frequency variations. The strange part is that the frequency does not sweep from ”-B/2 to B/2“ as expected.
[image]
[image]

Next, I connected the “File Sink” after synchronization and plotted the time-frequency spectrogram and LoRa signal. It is clearly observed that the LoRa signal now closely resembles the well-known LoRa signal. However, its frequency variations still appear somewhat unusual, and the time-frequency spectrogram no longer shows frequency changes, making it look quite strange.

[image]

[image]
This indicates that the received data is still incorrect. Where do you think the error might be? I know that Frame Sync can remove noise signals, but what exactly does this module do? What are its parameters? The device I am using is a Raspberry Pi 5. Could it be an issue with the Raspberry Pi?
What I need to do now is obtain correct LoRa signal data for subsequent deep learning training. If I connect the File Sink after the USRP, I will need to further process the data to remove the noise component.  

Looking forward to your response!

tapparelj, 2025-03-07:
I'm not surprised that the spectrogram can't show the chirps after synchronization (as the signal is not oversampled anymore, so your FFT is longer than the chirp duration.)
Try plotting the instantaneous frequency of the signal using something similar to unwrap(diff(angle(complex_samples))). You should be able to see the chirps again using this.

Remain123, 2025-03-10:
Thank you for your response. I used your suggestions and plotted the time-frequency spectrogram with the following code. 

[image]

According to formula

[image]

, I can calculate that the chirp is BW 125k.

My SF=7. This is the shape of the image I plotted for the first 128 points, and it can be seen that the frequency increases linearly. However, there is a gap in the middle. Could this be due to the fact that this might be a combination of parts from two preambles?

[image]

Plotting 128 points starting from the first point, but the image is not complete. This image seems to be half of the first preamble combined with half of another preamble. **I would like to know whether the output after synchronization includes the complete preamble.**

[image]