tapparelj/gr-lora_sdr #130: Problem in message decoded output

jasemanoah, 2025-02-20:
Hey, Can you help me out with this, I'm using a bladerf to receive rf using lora receiver but i get an error saying 'could not find port ascii in" also sometimes as "no message input".. 

[image]

jasemanoah, 2025-02-20:
[image]
I'm able to detect the transmitting signal when i disable lora receiver and view on the gui sink.

jasemanoah, 2025-02-25:
Yea, Thanks @tapparelj. I was able to solve the issue. Will this work to receive at SF = 5, The device I use is a BladeRF 2.0 xA4

[image]

tapparelj, 2025-02-25:
SF 5 is a bit special. The way it works is different based on the version of the LoRa chip you use.
What is the LoRa chip you use for the transmitter?

jasemanoah, 2025-02-25:
Oh is it. I shall let you know once i get that transmitter, probably in 2days. stay connected. Thanks for your response.

jasemanoah, 2025-02-27:
For SF 5 I'm able to detect the transmission on sink, But I get an error as "Header checksum invalid" and also the system crashes when running on SF 5. 

[image]

[image]

tapparelj, 2025-02-27:
The code from the main branch only works for SF5 with LoRa chips similar to sx1276. Those chips only allow implicit header mode for SF<7. If you use a chip that support SF5 with explicit header, you should use the branch _feature/sf5_6_sx126x_. 
Note that for now explicit sf5 (or 6) from that branch only works in the direction Semtech chip -> GNU Radio (and not the other way around)

jasemanoah, 2025-03-06:
Hey! Can LoRa decoders handle parallel streams?? As I want to capture multiple transmissions in the frequency range say, 430Mhz-470Mhz, So the transmission can be anywhere around this said frequency. I tried to building a custom block to frequency hop but couldn't capture all the packets, missed many of them(missed few and few broken packets). Since the LoRa transmission bandwidth is 500khz the Lora Rx is set to the same bandwidth, can we cover a 40Mhz band? Or is there any other way to not miss the packets? (Trying for SF-7 only)

tapparelj, 2025-03-07:
Hi,
I'm not sure what you tried to implement, but nothing prevents you from decoding parallel streams. 
However, I can't provide you an out of the box implementation as it depends a lot of the problem you are trying to solve.